Vonger/vocore2 #2: Sta mode is not working

artificerpi, 2018-07-22:
I've got following error, which keep output in the serial screen.
```
[  826.460057] MlmeEnqueueForRecv(): un-recongnized mgmt->subtype=4, STA-xx:xx:xx:xx:xx:xx
[  832.135500] ERROR!!! 
```
vocore2 has connected to the AP, but it seems that the packets cannot be correctly parsed.
Any workaround for STA+AP right now?

Vonger, 2018-07-22:
no such issue for me, how do you setting up?

artificerpi, 2018-07-23:
By `Setup AP + STA Mode guide` from http://vocore.io/v2.html, I've tried both compiling from openwrt source codes and flashing firmware http://vonger.cn/misc/vocore2/20170527V.bin, but they all given above error.

Vonger, 2018-07-23:
try http://vonger.cn/misc/vocore2/20180723V.bin

artificerpi, 2018-07-23:
Referred to this [issue](https://github.com/respeaker/get_started_with_respeaker/issues/94), I tried to power device with usb charger and solved this issue. Suggest to document the power supply requirements. Thx.

closed 2018-07-23 by artificerpi