Vonger/vocore2 #3: apcli0 cann't get ip after wifi reload

elgent, 2018-08-08:
Have anyone meet my problem as below?
After modify the wireless to ap+sta mode correctly, when first start up, it probably work well.
1. exec shell cmd "wifi up";
2. rescan to connect another ssid by luci
3. /etc/init.d/network restart
When i do any of above, the apcli0 can connect success, but can't get ip again.
If i reboot the device now or do the following steps, it probably work well again.
ifconfig apcli0 down
ifconfig ra0 down
rmmod mt7628
insmod mt7628
wifi up

Any suggestion? Thank you.

rkkoszewski, 2018-08-14:
Same here on a Vocore 2 Lite. Once it connects to an AP and then you restart the wifi, it won't get a new IP unless you restart the device.

Vonger, 2018-08-14:
currently solution is have to restart...
I have no idea about this bug yet :)

rkkoszewski, 2018-08-14:
From what I could observe, it doesn't seem to be an issue with the _ip_, but a connection issue.

If you run "iwinfo" while connected it shows the AP and the AP MAC. When you run "wifi" and wait a bit, the AP name will show up but the AP MAC will be 00:00:00...

Also if you set a static IP, you will see that you will get nothing trough.

elgent, 2018-08-15:
I found not every time 'rmmod + insmod mt7628' take effect now
This problem has puzzled me for a long time :(

elgent, 2018-08-15:
@rkkoszewski I agree with you
It seem that something is wrong in the driver cause fail to send pkt out...
But i have on idea now after review some driver code

rkkoszewski, 2018-11-07:
Any update on this? Has there been any work with this? There have been several commits and I'm unable to test it right now.

Vonger, 2019-02-17:
It should fixed by the new script.

rkkoszewski, 2019-02-17:
Issue is still present. Completely fresh build following Vocore2.git instructions as per Readme. I go to Luci, connect to an AP (Run ifconfig trough serial and you see an IP), and when I click on "restart" ra0, apcli0 never manages to get an IP again (Run ifconfig trough serial, and no IP in sight). Restart of Vocore2 Lite fixes the issue as always.

Vonger, 2019-02-18:
I just update new patch ralink.sh has updated.
This time it should be totally fixed :)

rkkoszewski, 2019-02-18:
I confirm, seems to be fixed now. I just compiled it again from scratch and flashed it on my Vocore 2 Lite. When you reconnect from command line with "wifi" it reconnects and gets the IP. If you use Luci and restart the wifi interface, it resets the wifi and finds an IP also. So it seems like the IP bug after wifi restart is fixed.

One thing I noticed is that when you ONLY disable the wifi client from Luci, the AP data gets copied to the Client and it breaks everything till you restart. But that's a different bug.

Vonger, 2019-02-24:
@rkkoszewski that might be luci bug.

closed 2019-02-24 by Vonger