Vonger/vocore2 #4: STA doesn't work when AP is disabled

rkkoszewski, 2018-08-14:
Hi, 

When I have AP setup and STA also connected everything seems fine. But when I set AP as disabled = 1 then STA will never connect.

iwinfo shows nothing at all, just as if the wifi chip is completely powered off. Setting AP disabled = 0, makes STA work again and connect properly. Why is this happening?

wifi up does nothing. But after setting AP disabled = 0 it inmediately works.

Vonger, 2018-08-15:
This is not a bug...
It is because this driver only supports AP and AP+STA mode.
Pure STA mode is supported by another driver which is also support AdHoc, but that driver do not support some AP features.
So my advance is to use hidden SSID, disable the board cast of AP SSID then it is working "like" a STA driver.

jsantala, 2018-08-15:
What driver supports AdHoc properly? The free mt76 driver has it, but it doesn't really work yet with VoCore2.

Vonger, 2018-08-15:
It is MT7628_LinuxSTA_V4.0.1.5_DPA_20160118.tar.bz2 in mediatek SDK for linux 3.14.
Currently we are using the alter driver from mediatek openwrt.

STA driver has a lot of code related closely to the kernel, I have no luck yet to hack it out like this AP driver. :(

closed 2019-02-17 by Vonger