Vonger/vocore2 #5: STA broken with SSID names that have spaces or "(" or ")"

rkkoszewski, 2018-08-14:
Hi,

I'm unable to connect to a WiFi network that has a space, "(" and ")" in its name (in STA mode). When I do iwinfo ra0 scan it shows all SSIDs till the first space. I tried removing the spaces from the SSID and I can connect properly. Using only BSSID doesn't work, as it seems to require a SSID, therefore any AP that has a space is inaccessible. 

Any idea what could be causing this issue?

elgent, 2018-08-15:
try this:
modify "iwpriv $ifname set ApCliSsid=$ssid" in ralink.sh to
iwpriv $ifname set "ApCliSsid=$ssid"

elgent, 2018-08-15:
...do iwinfo ra0 scan it shows all SSIDs till the first space
because output of iwinfo use space as delimiter, you can change it to an invisible char

rkkoszewski, 2018-08-15:
Nice! It works! Thanks!!

Created a pull request with the fix: https://github.com/Vonger/vocore2/pull/6

BTW, @elgent I could observe that "iwpriv ra0 get_site_survey" shows the SSIDs with full name "Inclusive Spaces", but in Luci Wireless scan it shows the names till the first space, breaking the Encryption and making the "Join Network" feature broken. Seems like Luci is using iwinfo for this part, do you know where you can change the spaces to make this work properly?

rkkoszewski, 2018-08-15:
Ok, so I found that this was implemented in the 080_iwinfo..patch file and implemented the separation of information relative to the columns instead of by space. Now APs with spaces work properly and the Luci Wireless scan interface works properly now.

Here is the PR: https://github.com/Vonger/vocore2/pull/7

closed 2018-08-15 by rkkoszewski