Vonger/vocore2 #10: SDHCI not found

james-tate, 2018-08-20:
I am getting the following error when building "Error: ../dts/VOCORE2.dts:114.2-3 label or path, 'sdhci', not found". 

I have tried enabling both  kmod-sdhci and kmod-sdhci-mt7620. What other dependencies are needed?

/home/cross/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7628/linux-3.18.29/scripts/dtc/dtc -O dtb -o /home/cross/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7628/VOCORE2.dtb ../dts/VOCORE2.dts
Error: ../dts/VOCORE2.dts:114.2-3 label or path, 'sdhci', not found
FATAL ERROR: Syntax error parsing input tree
Makefile:1077: recipe for target 'install' failed
make[5]: *** [install] Error 1
make[5]: Leaving directory '/home/cross/openwrt/target/linux/ramips/image'
Makefile:26: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/cross/openwrt/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/cross/openwrt/target/linux'
target/Makefile:21: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/cross/openwrt'
target/Makefile:17: recipe for target '/home/cross/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.target_install' failed
make[1]: *** [/home/cross/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/cross/openwrt'
Build failed - please re-run with -j1 to see the real error message
/home/cross/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 1

Vonger, 2018-08-26:
which version of openwrt you are using?
I find similar issue(not same, compile no problem), because openwrt update its sdhci driver.

james-tate, 2018-08-26:
Looks like in the newest openwrt version this is corrected. I was patching this driver into the openwrt source  from http://vonger.cn/misc/vocore2/openwrt.tar.bz2.

james-tate, 2018-08-26:
I do have a new problem with the current openwrt build for vocore2. I am using current https://github.com/openwrt/openwrt. The ttyS2 console stops after a few seconds (see behavior below). If I ssh in and try "echo hello >> /dev/ttyS2" I get nothing while connected to UART. Using the precompiled upgrade this works fine. I have peeked at the .dts files and I don't spot a problem right away. Any ideas?

---------------------------------------------------------------------------------------------------------
*** Warning - bad CRC, using default environment

============================================
Ralink UBoot Version: 4.3.0.0
--------------------------------------------
ASIC 7628_MP (Port5<->None)
DRAM component: 1024 Mbits DDR, width 16
DRAM bus: 16 bit
Total memory: 128 MBytes
Flash component: SPI Flash
Date:Jun 25 2016  Time:22:54:33
============================================
icache: sets:512, ways:4, linesz:32 ,total:65536
dcache: sets:256, ways:4, linesz:32 ,total:32768
RESET MT7628 PHY!!!!!!
Please choose the operation:
   0: Load system code then write to Flash via SERIAL.
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Entr boot command line interface.
   5: Load system code then write to Flash via USB Storage.
   7: Load Boot Loader code then write to Flash via Serial.
   8: Start Web Server to load system code.
   9: Load Boot Loader code then write to Flash via TFTP.
gpio trigger enabled
GPIO 46 used to trigger webpage
Hold button for between 3 seconds and 6 seconds
        then release to trigger webpage to load image
Hold button for between 6 seconds and 9 seconds
        then release to trigger loading image from USBSTORAGE
GPIO 46 is low                                                                                                                                  0

3: System Boot system code via Flash.
## Booting image at bc050000 ...
   Image Name:   MIPS OpenWrt Linux-4.14.66
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1545423 Bytes =  1.5 MB
   Load Address: 80000000
   Entry Point:  80000000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80000000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 4.14.66 (cross@cross) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7946-2c01425)) #0 Sun Aug 26 16:24:02 2018
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
[    0.000000] MIPS: machine is VoCore2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[... 74 lines cut ...]
[    0.572122] This architecture does not have kernel memory protection.
[    1.752450] init: Console is alive
[    1.756131] init: - watchdog -
[    2.260944] random: fast init done
[    3.040195] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    3.135398] usbcore: registered new interface driver usbfs
[    3.141143] usbcore: registered new interface driver hub
[    3.146636] usbcore: registered new device driver usb
[    3.157367] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    3.165751] ehci-platform: EHCI generic platform driver
[    3.181444] phy phy-10120000.usbphy.0: remote usb device wakeup disabled
[    3.188246] phy phy-10120000.usbphy.0: UTMI 16bit 30MHz
[    3.193572] ehci-platform 101c0000.ehci: EHCI Host Controller
[    3.199454] ehci-platform 101c0000.ehci: new USB bus registered, assigned bus number 1
[    3.207600] ehci-platform 101c0000.ehci: irq 26, io mem 0x101c0000
[    3.239243] ehci-platform 101c0000.ehci: USB 2.0 started, EHCI 1.00
[    3.246598] hub 1-0:1.0: USB hub found
[    3.250858] hub 1-0:1.0: 1 port detected
[    3.258242] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    3.266211] ohci-platform: OHCI generic platform driver
[    3.271851] ohci-platform 101c1000.ohci: Generic Platform OHCI controller
[    3.278765] ohci-platform 101c1000.ohci: new USB bus registered, assigned bus number 2
[    3.286930] ohci-platform 101c1000.ohci: irq 26, io mem 0x101c1000
[    3.364351] hub 2-0:1.0: USB hub found
[    3.368538] hub 2-0:1.0: 1 port detected

Vonger, 2018-08-30:
I have find this problem, it is openwrt commit 3a8efaef00d6b7317c46fe6b7e923912d8bf3500 broken the sdhci driver, make UART2 can not work properly. I have emailed openwrt but no reply yet. Please also email them to require revert that patch :)

james-tate, 2018-08-30:
Will do! Thanks for helping track it down.

closed 2018-08-30 by james-tate