Vonger/vocore2 #11: Help to install qmi on vocore2?

artificerpi, 2019-01-11:
I tried to use self-build openwrt 18.06.1 from source code, but got following error when I install qmi package:

```bash
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod:
 *      kernel (= 4.14.63-1-6c5df541905ae6bbe8d85627558ea00c) *   
```

With `uname -r`  I got:
```
4.14.63
```

Why does the kernel version not match? BTW, does vocore2 support USB LTE modem?

Vonger, 2019-01-11:
This is openwrt version do not match , normally happens when you make from trunk.
You can force install it or modify it suffix to zip and uncompress then insmod manually.
VoCore2 support USB 4G well.

artificerpi, 2019-01-11:
The source code I use is from https://github.com/openwrt/openwrt/archive/v18.06.1.tar.gz. I've tried to force install it, but after installation, vocore2 keeps rebooting over and over again. Is there a required order to install lte modem related packages?

artificerpi, 2019-01-13:
@Vonger I've succeed in installing all qmi LTE related packages, but got following error:

```
[ 2058.200217] usb 1-1: usbfs: interface 1 claimed by usbfs while 'usbmode' sets config #0

[ 2058.508765] usb 1-1: usbfs: process 20769 (usbmode) did not claim interface 1 before use
```

Have you met similar issue or do you have any idea how to fix it?

The LTE modem device I'm using is Huawei ME909s-821. 

Thanks!