Vonger/vocore2 #12: Failed patch against openwrt master

brainstorm, 2019-02-17:
```
vagrant@ubuntu-cosmic:/home/ubuntu/openwrt$ patch -p1 < ./package/vocore2/mt7628/openwrt/000-*.patch
patching file target/linux/ramips/patches-4.14/801-mt7628-kernel-wext-oldapi.patch
patching file package/base-files/files/etc/shadow
patching file target/linux/ramips/dts/VOCORE2.dts
Hunk #1 succeeded at 64 (offset 11 lines).
patching file package/base-files/files/etc/config/fstab
patching file target/linux/ramips/patches-4.14/0306-spi-flash-factory-id.patch
patching file target/linux/ramips/patches-4.14/882-pinctrl-fix-refclk-name.patch
patching file target/linux/ramips/patches-4.14/0307-suppress-mmc-warning.patch
patching file package/system/opkg/files/customfeeds.conf
patching file target/linux/ramips/base-files/etc/board.d/01_leds
Hunk #1 FAILED at 447.
1 out of 1 hunk FAILED -- saving rejects to file target/linux/ramips/base-files/etc/board.d/01_leds.rej
patching file package/base-files/files/bin/config_generate
patching file target/linux/ramips/base-files/etc/board.d/02_network
Hunk #1 succeeded at 398 with fuzz 1 (offset 49 lines).
```

brainstorm, 2019-02-17:
This patch should fix it instead:

```
--- 01_leds.orig	2019-02-17 00:27:07.299273307 +0000
+++ 01_leds	2019-02-17 00:37:29.699273307 +0000
@@ -380,6 +380,7 @@
 vocore-8M|\
 vocore-16M)
 	ucidef_set_led_netdev "eth" "ETH" "vocore:orange:eth" "eth0"
+	set_wifi_led "$boardname:green:status" "ra0"
 	;;
 w502u)
 	set_wifi_led "rt2800pci-phy0::radio"
```

Vonger, 2019-02-17:
vocore-16M that is vocore one. not vocore2

brainstorm, 2019-02-17:
Thanks for the fast response! That's weird, there seems to be no pre-existing `VoCore2` case-entry on trunk (HEAD) for that file, I'll add one then, thanks!

I would love to see those patches upstream though on HEAD, are most of the hunks on this repo included on openwrt HEAD?

Vonger, 2019-02-17:
ye, it isweird, looks like somebody remove it from trunk, but openwrt 18.06.2 have it.

closed 2019-02-24 by Vonger