Vonger/vocore2 #13: Luci patches do not apply against HEAD

brainstorm, 2019-02-17:
Luci patches seem outdated:

```
$ patch -p1 < ./package/vocore2/mt7628/openwrt/luci/*.patch
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/feeds/luci/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua
|+++ b/feeds/luci/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua
--------------------------
File to patch: ^C
```

There's no `model` folder anymore, but just `controller`. Moreover `wifi.lua` seems to have moved elsewhere:

```
$ find feeds/luci -iname "wifi.lua"
feeds/luci/modules/luci-mod-network/luasrc/model/cbi/admin_network/wifi.lua
feeds/luci/modules/luci-mod-admin-mini/luasrc/model/cbi/mini/wifi.lua
```

Vonger, 2019-02-17:
the patch is for OpenWrt 18.06.2, not trunk.

brainstorm, 2019-02-17:
I see several references to VoCore2 on HEAD (trunk) already... is it safe to just stick with openwrt HEAD or are the patches in this repo crucial to have a working wifi?

brainstorm, 2019-02-17:
@Vonger, don't get me wrong, this work is awesome and I think it would be beneficial to have a [draft pullrequest](https://github.blog/2019-02-14-introducing-draft-pull-requests/) opened against openwrt master so that other devs could contribute against it just by directly pulling from your (ongoing) branch. 

Working with patches is tedious for everyone involved vs branches and pullrequests, imho.

Cheers!

Vonger, 2019-02-17:
@brainstorm Yes, I think this help too, but openwrt do not allow such patch submit to it. It is not compatible with 80211 driver.
If you need to use it in trunk version, need some fix in the patch.

brainstorm, 2019-02-17:
Thanks for the insight, @Vonger! Is the upstream 80211 functional (as a mere client) or does it have major defects we should be aware of?

Vonger, 2019-02-17:
the driver from mediatek do not support mac80211, it is using some old way ioctl and iwpriv...actually we can add patch to make it support mac80211, but need a lot of work.

closed 2019-02-24 by Vonger