Vonger/vocore2 #14: es8388 record no voice

lloovve, 2019-03-26:
i use this code, play wav file is ok,but record no voice? need help

Vonger, 2019-04-04:
this bug is because es8328 driver did not update register 0x03 correctly. microphone bias did not enabled. once we record audio it change 0x03 from 0xfc to 0x0c but it should be 0x03.
I will looking into this later.

lloovve, 2019-04-06:
i try this work for few days  ,maybe this bug we shuld work no register ES8328_DACCONTROL21(reg 43) slrck  lrck_sel     ES8328_ADCCONTROL3 (reg 11)  TRI

Vonger, 2019-04-06:
I use a simple logic analyser do this. 
I tried your register, looks like it work more or less, at least switch is all on, but noise output it should be some volume control is not turned to max

lloovve, 2019-04-07:
https://github.com/espressif/esp-adf/blob/1dd8a6057d9a70ed4b70bda83fed8b10528fbb05/components/audio_hal/driver/es8388/es8388.c

http://dl.radxa.com/rock2/docs/hw/ds/ES8388%20user%20Guide.pdf

maybe help

Vonger, 2019-04-07:
actually once I make it work, 20170527 version, but not using es8328 driver, directly hard coded into driver.

03 0c -> 03 00
0d 22 -> 0d 33(272)
18 22 -> 18 33
19 24 -> 19 20

After fix the upper register, it works
But es8328 driver modify the registers to default...so still need to code and understand the driver to make it work ;)

lloovve, 2019-04-07:
[https://0d077ef9e74d8.cdn.sohucs.com/rmV9rFz_png](url)

I noticed that  es8388 datasheet ,slave mode not support 272 Ratio ,and es8388 in slave mode will auto detect MCLK/LRCK ratio.

lloovve, 2019-04-07:
[image]

lloovve, 2019-04-07:
i change es8388 as master mode and 7688 slaver mode rec is ok, so the problem is above

all set

                / / //复位所有寄存器
		snd_soc_write(codec, ES8328_CONTROL1, 0x80);
		//MCLK 分频系数设置
		snd_soc_write(codec, ES8328_CONTROL1, 0x75);
		//停止状态机 DLL
		snd_soc_write(codec, ES8328_CHIPPOWER, 0xF3);
		//主机模式
		snd_soc_write(codec, ES8328_MASTERMODE, 0x80);
		//DAC slrck 控制 same LRCK
		snd_soc_write(codec, ES8328_DACCONTROL21, 0x80);
		//reg 0  参考电压
		snd_soc_write(codec, ES8328_CONTROL1, 0x05);//75
		//reg 1  参考电压
		snd_soc_write(codec, ES8328_CONTROL2, 0x40);
		//ADC 电源管理  全部开启
		snd_soc_write(codec, ES8328_ADCPOWER, 0x00);
		//DAC 电源开启
		snd_soc_write(codec, ES8328_DACPOWER, 0x00);
		//ADC PGA控制增益
		snd_soc_write(codec, ES8328_ADCCONTROL1, 0x7F);
		//ADC 路径 LR 差分输入
		snd_soc_write(codec, ES8328_ADCCONTROL2, 0xF0);
		//ADC 路径 LINPUT1-RINPUT1 差分引脚对 立体声  ASDOUT  输出
		snd_soc_write(codec, ES8328_ADCCONTROL3, 0x00);
		//ADC control4 ADC数据格式 i2s 数据格式 ADC数据长度 
		//全部为左声道数据 16bit  i2s格式
		snd_soc_write(codec, ES8328_ADCCONTROL4, 0x0C);
		//reg 24  DACFsRatio=272
		snd_soc_write(codec, ES8328_DACCONTROL2, 0x13);
		//ADC control5  MCLK 控制 双速模式ADCFsRatio=13   272
		snd_soc_write(codec, ES8328_ADCCONTROL5, 0x13);
		//ADC daigial 数字音量 L 左声道 ADC control 9 
		snd_soc_write(codec, ES8328_ADCCONTROL8, 0x00);
		//ADC daigial 数字音量 R 右声道 ADC control 9 
		snd_soc_write(codec, ES8328_ADCCONTROL9, 0x00);
		//ADCRampRate=0.5 ADCSoftRamp=1 ADCLeR=1 ADCMute=0
		snd_soc_write(codec, 0x0F, 0x30);//ADC音量0x30   0x04 静音ADC

		//参考电压 DAC = off ADC =on DAC DLL=off ADC DLL= on  STM DAC =off STM ADC = on 
		//ADC DEM = on
		//DAC DEM = off
		snd_soc_write(codec, ES8328_CHIPPOWER, 0x00);

lloovve, 2019-04-07:
[image]

lloovve, 2019-04-07:
DTS file ,i am not sure 

sound {
		compatible = "simple-audio-card";
		simple-audio-card,name = "soundcard";
		simple-audio-card,format = "i2s";
		**simple-audio-card,bitclock-master = <&sound_master>;
		simple-audio-card,frame-master = <&sound_master>;**
		simple-audio-card,widgets =
			"Microphone", "Microphone Jack",
			"Headphone", "Headphone Jack";
		simple-audio-card,routing =
			"LINPUT1", "Microphone Jack",
			"RINPUT1", "Microphone Jack",
			"Headphone Jack", "LOUT1",
			"Headphone Jack", "ROUT1";
		simple-audio-card,mclk-fs = <272>;

		simple-audio-card,cpu {
			sound-dai = <&i2s>;
		};

		sound_master:simple-audio-card,codec {
			sound-dai = <&codec>;
		};
	};

Vonger, 2019-04-12:
not that problem. I can record normally now, later I will submit patch.

Vonger, 2019-04-16:
emm, still have noise use master mode..

Vonger, 2019-04-22:
works normal now, closed

closed 2019-04-22 by Vonger