Vonger/vocore2 #17: Support version 19.0.x?

uniformsierra, 2020-07-08:
Hi, not sure if this is question or a issue... 
VoCore2 - thank you for a great fun board, I am learning a lot as I use it.

Will these patches support OpenWRT 19.0.7?
I need to run kernel 5.4 or higher with the i2c and microSD card, but these cannot work.

I can build and install a default image using OpenWRT, but cannot get i2c or microSD working.

It seems these patches are necessary, or something with /dev/mem driver like Omega (https://innomatic.home.blog/tag/mt7688/)? But these patches don't work.
I am beginner so don't really know.

Vonger, 2020-07-10:
Working on this...it will take some time, a lot of driver need to port

Vonger, 2020-07-29:
80% supported, close for now :)

closed 2020-07-29 by Vonger