Vonger/vocore2 #19: no sdcard device on OpenWRT 19.07.5

sbrinkmann, 2020-12-28:
I followed your instructions to build OpenWRT 19.07 on my own. Everything works well, but unfortunally the sd card device /dev/mmcblk0 doesn't show up in the list of available devices. When I install the firmware http://vonger.cn/misc/vocore2/20200812.19073.bin the sdcard device shows up and works flawlessly. I've compared the installed kernel mods from your to my image and they're identical, up to some extra mods which I've added.

```
BusyBox v1.30.1 () built-in shell (ash)

    _    __        ______                   ___
   | |  / /____   / ____/____   _____ ___  |__ \
   | | / // __ \ / /    / __ \ / ___// _ \ __/ /
   | |/ // /_/ // /___ / /_/ // /   /  __// __/
   |___/ \____/ \____/ \____//_/    \___//____/

 -----------------------------------------------------
 OpenWrt 19.07.5, r11257-5090152ae3
 -----------------------------------------------------
root@OpenWrt:~# lsmod
cfg80211              233812  3 mt7603e,mt76,mac80211
compat                  6519  2 mac80211,cfg80211
crc_ccitt               1035  1 ppp_async
crc16                   1031  1 ext4
crc32c_generic          1424  0
crypto_hash            10066  3 ext4,jbd2,crc32c_generic
ehci_hcd               34871  1 ehci_platform
ehci_platform           4656  0
evdev                  10096  0
exfat                 103995  0
ext4                  383189  0
fat                    52638  1 vfat
fbusb                  10368  0
gpio_button_hotplug     6896  0
hid                    82550  2 usbhid,hid_generic
hid_generic              624  0
i2c_core               22728  2 i2c_mt7621,i2c_dev
i2c_dev                 4576  0
i2c_mt7621              2480  0
ip_tables              10061  4 iptable_nat,iptable_mangle,iptable_filter
ip6_tables              9825 24 ip6table_mangle,ip6table_filter
ip6t_REJECT             1072  2
ip6table_filter          688  1
ip6table_mangle         1232  1
ipt_MASQUERADE           720  2
ipt_REJECT              1008  2
iptable_filter           688  1
iptable_mangle           944  1
iptable_nat              752  1
jbd2                   52801  1 ext4
leds_gpio               2928  0
ledtrig_usbport         2784  0
mac80211              464552  2 mt7603e,mt76
mbcache                 3182  1 ext4
mmc_block              23654  0
mmc_core               87477  2 mmc_block,mtk_sd
mt76                   33068  1 mt7603e
mt7603e                38636  0
mtk_sd                 20192  0
nf_conntrack           57893 12 nf_conntrack_ipv6,ipt_MASQUERADE,xt_state,xt_nat,xt_conntrack,xt_REDIRECT,nf_nat_masquerade_ipv4,nf_conntrack_ipv4,nf_nat_ipv4,nf_nat,nf_flow_table,nf_conntrack_rtcache
nf_conntrack_ipv4       4368 16
nf_conntrack_ipv6       4576  7
nf_conntrack_rtcache    2672  0
nf_defrag_ipv4          1046  1 nf_conntrack_ipv4
nf_defrag_ipv6          4798  1 nf_conntrack_ipv6
nf_flow_table          13791  2 xt_FLOWOFFLOAD,nf_flow_table_hw
nf_flow_table_hw        2192  1
nf_log_common           2703  2 nf_log_ipv4,nf_log_ipv6
nf_log_ipv4             3472  0
nf_log_ipv6             4176  0
nf_nat                  9451  4 xt_nat,nf_nat_redirect,nf_nat_masquerade_ipv4,nf_nat_ipv4
nf_nat_ipv4             3441  1 iptable_nat
nf_nat_masquerade_ipv4    1756  1 ipt_MASQUERADE
nf_nat_redirect         1531  1 xt_REDIRECT
nf_reject_ipv4          2147  1 ipt_REJECT
nf_reject_ipv6          2536  1 ip6t_REJECT
nls_base                5152  7 vfat,fat,nls_utf8,nls_iso8859_1,nls_cp437,exfat,usbcore
nls_cp437               4432  0
nls_iso8859_1           2896  0
nls_utf8                 880  0
ohci_hcd               22850  1 ohci_platform
ohci_platform           4176  0
ppp_async               6928  0
ppp_generic            22786  3 pppoe,ppp_async,pppox
pppoe                   9232  0
pppox                   1386  1 pppoe
scsi_mod               93074  2 usb_storage,sd_mod
sd_mod                 29728  0
slhc                    5307  1 ppp_generic
tun                    22353  0
usb_common              2551  1 usbcore
usb_storage            39675  0
usbcore               132579  8 usbhid,ledtrig_usbport,fbusb,usb_storage,ohci_platform,ohci_hcd,ehci_platform,ehci_hcd
usbhid                 23530  0
vfat                    8656  0
x_tables               13999 23 ipt_REJECT,ipt_MASQUERADE,xt_time,xt_tcpudp,xt_state,xt_nat,xt_multiport,xt_mark,xt_mac,xt_limit,xt_conntrack,xt_comment,xt_TCPMSS,xt_REDIRECT,xt_LOG,xt_FLOWOFFLOAD,iptable_mangle,iptable_filter,ip_tables,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xt_FLOWOFFLOAD          3072  0
xt_LOG                   816  0
xt_REDIRECT              752  0
xt_TCPMSS               2832  4
xt_comment               528164
xt_conntrack            2352 20
xt_limit                1232 20
xt_mac                   656  0
xt_mark                  720  0
xt_multiport            1328  0
xt_nat                  1584  0
xt_state                 752  0
xt_tcpudp               1840 12
xt_time                 1808  0
```

sbrinkmann, 2020-12-28:
I was able to get 19.07.5 running smoothly including sd card support. Everything was there but the _README.md_ for 19.07 is unsuffient. I've added these two steps, then it worked like a charme:

1. copy config-4.14 to openwrt kernel config folder to avoid missing any kernel module.
```
cd path/to/openwrt
cp ./package/vocore2/openwrt.1907/config-4.14 ./target/linux/ramips/mt76x8/
```

2. apply patch to enable sd card
`patch -p1 < ./package/vocore2/openwrt.1907/vocore2-default-enable-sdcard.patch`

Vonger, 2021-11-06:
done :)

closed 2021-11-06 by Vonger