Vonger/vocore2 #20: 2 microphones (dual/stereo sound recording)

dumblob, 2021-08-14:
Newbie here, bare with me.

How could I connect two microphones to vocore2?

I read on the internet about connecting 1 microphone, but couldn't find any info about 2 microphones (to get real stereo sound) simultaneously :cry:.

If it's impossible, could you recommend some very thin (1-3mm) HAT offering pins for soldering of 2 microphones?

Vonger, 2021-11-06:
it can support two microphone, but you have to modify the driver setting also do some hardware hack, not easy for newbie.

closed 2021-11-06 by Vonger

dumblob, 2021-11-06:
That's a pity it's quite some (non-trivial) work.

Is the hardware "hack" you speak about not "upstreamable" (i.e. it would prevent some other functionality and thus it won't be done in next HW revisions)?

And assuming we did the HW hack and modified the driver, would the driver modifications be accepted upstream? Are they backwards-compatible with the one-microphone setup?