Vonger/vocore2 #21: Can't rotate with raspberry

ichbaum, 2021-09-06:
Hi,
i have trouble to rotate the display with X.
I'm using a Raspberry Pi 3 with MainsailOs (Raspbian GNU/Linux 10 (buster)).
The Display works fine without the rotate option in 99-fbturbo.conf.
But I can't use it whith the option.  

startx shows the error "undefined symbol: xf86DisableRandR".

```
pi@mainsailos:~ $ sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
```
Section "Device"
        Identifier      "LCD"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"
        **Option          "Rotate" "CCW"**
        Option          "SwapbuffersWait" "true"
EndSection

```
pi@mainsailos:~ $ sudo FRAMEBUFFER=/dev/fb1 startx

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.10.17-v8+ aarch64 Raspbian
Current Operating System: Linux mainsailos 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=tty1 root=PARTUUID=6ed37222-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 19 April 2021  03:23:36PM
xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Mon Sep  6 08:24:13 2021
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.60-v7+
/usr/lib/xorg/Xorg: symbol lookup error: /usr/lib/xorg/modules/drivers/fbturbo_drv.so: undefined symbol: xf86DisableRandR
```

```
[   191.646] (DB) xf86MergeOutputClassOptions unsupported bus type 0
[   191.646] (**) FBTURBO(0): Option "Rotate" "CCW"
[   191.646] (**) FBTURBO(0): Option "fbdev" "/dev/fb1"
[   191.646] (**) FBTURBO(0): Option "SwapbuffersWait" "true"
[   191.646] (II) FBTURBO(0): processor: Unknown
[   191.646] (**) FBTURBO(0): rotating screen counter-clockwise
[   191.646] (II) FBTURBO(0): checking modes against framebuffer device...
[   191.646] (II) FBTURBO(0): checking modes against monitor...
[   191.646] (II) FBTURBO(0): Virtual size is 480x800 (pitch 480)
[   191.646] (**) FBTURBO(0):  Built-in mode "current"
[   191.647] (==) FBTURBO(0): DPI set to (96, 96)
[   191.647] (II) Loading sub module "fb"
[   191.647] (II) LoadModule: "fb"
[   191.647] (II) Loading /usr/lib/xorg/modules/libfb.so
[   191.647] (II) Module fb: vendor="X.Org Foundation"
[   191.647]    compiled for 1.20.4, module version = 1.0.0
[   191.647]    ABI class: X.Org ANSI C Emulation, version 0.4
[   191.647] (**) FBTURBO(0): using shadow framebuffer
[   191.647] (II) Loading sub module "shadow"
[   191.647] (II) LoadModule: "shadow"
[   191.648] (II) Loading /usr/lib/xorg/modules/libshadow.so
[   191.652] (II) Module shadow: vendor="X.Org Foundation"
[   191.652]    compiled for 1.20.4, module version = 1.1.0
[   191.652]    ABI class: X.Org ANSI C Emulation, version 0.4
[   191.652] (II) FBTURBO(0): FBIOBLANK: Invalid argument (Screen blanking not supported by kernel - disabling)
[   191.662] (II) FBTURBO(0): can't load 'g2d_23' kernel module
[   191.663] (II) FBTURBO(0): failed to enable the use of sunxi display controller
[   191.663] (II) FBTURBO(0): No sunxi-g2d hardware detected (check /dev/disp and /dev/g2d)
[   191.663] (II) FBTURBO(0): G2D hardware acceleration can't be enabled
[   191.663] (II) FBTURBO(0): enabled VFP/NEON optimizations
[   191.663] (II) FBTURBO(0): display rotated; disabling DGA
[   191.663] (II) FBTURBO(0): using driver rotation; disabling XRandR

```

My installation steps.
Works without the rotate.
```
#Update system and kernel
sudo apt-get update
sudo apt-get upgrade
sudo reboot
#install kernel-headers
sudo apt install raspberrypi-kernel-headers

#get vocore 
cd /tmp
wget https://github.com/Vonger/vocore2/archive/refs/heads/master.zip
unzip master.zip
cd /tmp/vocore2-master/utils/fbusb/src
#make vocore driver
make -C /usr/src/linux-headers-$(uname -r ) M=$PWD modules
sudo insmod fbusb.ko
sudo cp fbusb.ko /lib/modules/$(uname -r)/kernel/drivers/

#Add fbusb.ko to load.d
sudo nano /etc/modules-load.d

#1. set screen to black: 
dd if=/dev/zero of=/dev/fb1 bs=1000 count=768

#2. set screen to random pixel: 
dd if=/dev/urandom of=/dev/fb1 bs=1000 count=768

#https://klipperscreen.readthedocs.io/en/latest/Installation/
#KlipperScreen Installation
cd ~/
git clone https://github.com/jordanruthe/KlipperScreen.git
cd ~/KlipperScreen
./scripts/KlipperScreen-install.sh

#change from fb0 to fb1
sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf

Section "Device"
        Identifier      "LCD"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"
        Option          "Rotate" "CCW"
        Option          "SwapbuffersWait" "true"
EndSection

```

ichbaum, 2021-09-08:
Another test.
```
pi@mainsailos:~ $ sudo DISPLAY=:0 xrandr --output default --rotate left
xrandr: output default cannot use rotation "left" reflection "none"
```

ichbaum, 2021-09-09:
I found this.
A patch for X server to add xf86DisableRandR as Dummy for drivers that still use ist but shouldn't.

```
The patch removed function xf86DisableRandR. This patch re-adds it as a dummy function
that does nothing to fix linking with drivers that expect it. Drivers should no longer
call it, so putting it back empty has the same effect.
```

https://build.opensuse.org/package/view_file/openSUSE:Leap:15.0/xorg-x11-server/n_add-dummy-xf86DisableRandR.patch?expand=0

ichbaum, 2021-09-09:
Found a solution.

Use fbdev instead of fbturbo.
```

sudo rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf
sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf

Section "Device"
	Identifier "LCD"
	Driver "fbdev"
	Option "fbdev" "/dev/fb1"
	Option "Rotate" "CW"
	Option "SwapbuffersWait" "true"
EndSection

```

Vonger, 2021-11-06:
Hi, do you make it work with RPi now? I did  not tried this, maybe a interesting project.

ichbaum, 2021-11-06:
I will post a working configuration and install script later.
It works with the fbdev.

ichbaum, 2021-11-15:
```
#Update system and kernel
sudo apt-get update
sudo apt-get upgrade -y
sudo apt-get full-upgrade -y
sudo apt-get clean
sudo apt-get autoremove -y
#Reboot to use the possibly new kernel
sudo reboot
#install kernel-headers
sudo apt install raspberrypi-kernel-headers
```

```
#get vocore 
cd /tmp
wget https://github.com/Vonger/vocore2/archive/refs/heads/master.zip
unzip master.zip
cd /tmp/vocore2-master/utils/fbusb/src
#make vocore driver
make -C /usr/src/linux-headers-$(uname -r ) M=$PWD modules
sudo insmod fbusb.ko
sudo cp fbusb.ko /lib/modules/$(uname -r)/kernel/drivers/

#Load driver at startup
sudo /sbin/depmod -a
```

```
#Test the screen
#1. set screen to black: 
sudo dd if=/dev/zero of=/dev/fb1 bs=1000 count=768

#2. set screen to random pixel: 
sudo dd if=/dev/urandom of=/dev/fb1 bs=1000 count=768
```

```
# install xorg driver
sudo apt-get install xserver-xorg-video-fbdev

#Set Console output to LCD
#add "fbcon=map:10 fbcon=rotate 1" to cmdline.txt
sudo nano /boot/cmdline.txt

#Test console
export FRAMEBUFFER=/dev/fb1
sudo FRAMEBUFFER=/dev/fb1 startx

```
```
#remove fbturbo and use fbdev
sudo rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf
```

```
sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf
#Add this text and save
Section "Device"
	Identifier "LCD"
	Driver "fbdev"
	Option "fbdev" "/dev/fb1"
	Option "Rotate" "CW"
	Option "SwapbuffersWait" "true"
EndSection

```
```
#Rotate the touch input too
sudo nano /etc/udev/rules.d/51-touchscreen.rules
#add to the file
ACTION=="add", ATTRS{name}=="VoCore Touch", ENV{LIBINPUT_CALIBRATION_MATRIX}="0 1 0 -1 0 1 0 0 1"
```

If the raspberry gets a new kernel, we need to recompile the driver.

```
#get vocore 
cd /tmp
wget https://github.com/Vonger/vocore2/archive/refs/heads/master.zip
unzip master.zip
cd /tmp/vocore2-master/utils/fbusb/src
#make vocore driver
make -C /usr/src/linux-headers-$(uname -r ) M=$PWD modules
sudo insmod fbusb.ko
sudo cp fbusb.ko /lib/modules/$(uname -r)/kernel/drivers/

#Load driver at startup
sudo /sbin/depmod -a
```