Vonger/vocore2 #27: no sound device (22.03.0)

basiliscos, 2022-09-10:
Hi,

I have compiled the openwrt with 

```
CONFIG_PACKAGE_kmod-sdhci-mt7620=y
CONFIG_PACKAGE_kmod-sound-mt7620=y
```

but there are no `/dev/mixer` etc. after reboot. What can be done?

newluhux, 2022-09-26:
new patch available:

https://github.com/Vonger/vocore2/pull/30

qtopie, 2022-11-20:
kmod-sound-mt7628 is not found in 22.03 but it's available in 21.02. 

[image]

BTW, I've tried and managed to make audio of my vocore2u work with firmware 21.02.

Vonger, 2023-02-20:
22.03 is using new kenerl of Linux, it has a weird bug will crash the driver, we are looking into this.

basiliscos, 2023-03-04:
any news so far?

Vonger, 2024-04-15:
> any news so far?

works now, apply the 23.05 sound patch

closed 2024-05-29 by Vonger