Vonger/vocore2 #43: Cannot Rotate FrameBuffer

Spacefreak18, 2023-05-20:
I compiled the fbusb module, and I have connected a vocore display to my amd64 PC.

However, no matter what I try, the screen is in portrait mode. I would like to achieve a clockwise 90 degrees rotation.

more /sys/class/graphics/fb1/rotate    gives a value of 0. And if i try to 
echo 1 > /sys/class/graphics/fb1/rotate   it doesn't take, it remains 0.

I am positive the device is attached as fb1.

Is there anyway to acheive this in hardware?

ieiao, 2024-08-02:
Hi, maybe you can use the new [drm driver](https://github.com/ieiao/mpro_drm.git) instead of old fbused driver.

In my case, with this driver, I can use the MPRO screen as a extension screen on debian bookworm (Intel integrated graphics and KDE desktop enviroument), and I can set its rotation to any direction.

You should chooise the properly repository branch according your kernel version, [this branch](https://github.com/ieiao/mpro_drm/tree/6.6.y-dma) provides the usage on Raspberry, amd64 also same.

Spacefreak18, 2024-08-07:
> Hi, maybe you can use the new [drm driver](https://github.com/ieiao/mpro_drm.git) instead of old fbused driver.
>
> In my case, with this driver, I can use the MPRO screen as a extension screen on [...]

nice work! i'll give it a shot

closed 2025-01-05 by Vonger