Vonger/vocore2 #52: fbusb compilation errors for Raspberry Pi Zero 2 W (Bookworm lite)

jbird60, 2024-05-28:
Attempting to compile fbusb as per [the guide](https://vonger.cn/?p=15349) leads to make errors.

```
carpi@carpi:~/vocore2/utils/fbusb/src $ make -C /usr/src/linux-headers-`uname -r`/ M=`pwd` modules # compile
make: Entering directory '/usr/src/linux-headers-6.6.28+rpt-rpi-v8'
  CC [M]  /home/carpi/vocore2/utils/fbusb/src/fbusb.o
/home/carpi/vocore2/utils/fbusb/src/fbusb.c: In function ‘fbusb_pause_show’:
/home/carpi/vocore2/utils/fbusb/src/fbusb.c:300:31: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘long int’ [-Wformat=]
  300 |         return sprintf(buf, "%d\n", uinfo->pause);
      |                              ~^     ~~~~~~~~~~~~
      |                               |          |
      |                               int        long int
      |                              %ld
/home/carpi/vocore2/utils/fbusb/src/fbusb.c: In function ‘fbusb_probe’:
/home/carpi/vocore2/utils/fbusb/src/fbusb.c:479:23: error: ‘FBINFO_DEFAULT’ undeclared (first use in this function)
  479 |         info->flags = FBINFO_DEFAULT | FBINFO_VIRTFB;
      |                       ^~~~~~~~~~~~~~
/home/carpi/vocore2/utils/fbusb/src/fbusb.c:479:23: note: each undeclared identifier is reported only once for each function it appears in
/home/carpi/vocore2/utils/fbusb/src/fbusb.c: In function ‘fbusb_pause_store’:
/home/carpi/vocore2/utils/fbusb/src/fbusb.c:308:9: warning: ignoring return value of ‘kstrtol’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  308 |         kstrtol(buf, 10, &uinfo->pause);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-headers-6.6.28+rpt-common-rpi/scripts/Makefile.build:248: /home/carpi/vocore2/utils/fbusb/src/fbusb.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.6.28+rpt-common-rpi/Makefile:1938: /home/carpi/vocore2/utils/fbusb/src] Error 2
make: *** [/usr/src/linux-headers-6.6.28+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.6.28+rpt-rpi-v8'
carpi@carpi:~/vocore2/utils/fbusb/src $ 
```

Device is a Raspberry Pi Zero 2 W, distro is 64-bit Pi OS lite (Bookworm).

Vonger, 2024-05-29:
Seems like the new kernel changed define. need to patch.
Also you can try our DRM driver.

ieiao, 2024-08-02:
Hi, currently MPRO screen(Vocore2 Display) has provided DRM driver support, which is easier to use than fb driver. You can get the latest driver and usage from [here](https://github.com/ieiao/mpro_drm/tree/6.6.y-dma).

To use this driver properly, please make sure your screen firmware already supports the partially draw.

closed 2024-08-02 by Vonger