Vonger/vocore2 #53: Vocore Dispaly turns white and black after the second

Grgur00, 2024-06-16:
Hi, I followed the steps you wrote on your blog: https://vonger.cn/?p=15349 I can send data to it eg make pixels random colours or black using dd if=/dev/urandom of=/dev/fb1 bs=1000 count=768 comand, but I can't seem to make it a functioning display. If I remove HDMI before booting nothing will be displayed. Please advise

Grgur00, 2024-06-16:
I manged to get it working hower different problem occurs I can't get it to display tty console also it dosent work if I dont connect HDMI to it here is my 99-fbturbo.conf
Section "Device"
        Identifier      "LCD"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"
        Option          "Rotate" "CCW"
        Option          "SwapbuffersWait" "true"
EndSection

ieiao, 2024-08-02:
Hi, currently MPRO screen(Vocore2 Display) has provided DRM driver support, which is easier to use than fb driver. You can get the latest driver and usage from [here](https://github.com/ieiao/mpro_drm/tree/6.6.y-dma).

To use this driver properly, please make sure your screen firmware already supports the partially draw.

closed 2025-01-05 by Vonger